digiblur/Tasmota #1: Martin Jerry Dimmer Button's don't work

33-viper-33, 2019-01-29:
I'm having a problem with the physical buttons using your code.  I've flashed with Tasmota **6.3.0.16(digiDIMv2) and copied and pasted the following code into Termite.

backlog** rule1 on switch3#state=2 do dimmer + endon on switch2#state=2 do dimmer - endon on switch2#state=3 do dimmer 20 endon on switch3#state=3 do dimmer 100 endon; rule1 1; setoption32 10; switchmode1 6; switchmode2 5; switchmode3 5

rule2 on switch1#state=3 do publish Table-Dimmer/Main TOGGLE endon rule2 1

No I didn't modify rule 2 yet to match my configuration.  When I press the physical buttons, I get a flash of LEDs only while the button is depressed, but no state change.  When I press and hold, I can see the state change in the web GUI from off to on, but it reverts when button is released.  There is no state change when trying to turn off, even while being held but i do see both red and blue lights illuminated.  

I'm not sure where to pull the state information since there is no console in this version of tasmota to watch the commands as they are taking place nor does it appear to be reporting in termite.

Any help is greatly appreciated.

digiblur, 2019-01-29:
If there isn't a web console button to watch the commands and such then you either have the wrong firmware or a bad firmware flash as it is standard Tasmota with just a few minor changes.

33-viper-33, 2019-01-29:
Ok, I re-downloaded and re-flashed from a different device using the ESPtool this time vs the web interface.  I now have a console and can watch the states.  I coppied rule1 from above again but did not copy rule2 into the web GUI console.  I'm receiving the same results with button presses.  Below is the console output from 2 button presses.  You will se it turn on when pressed and off when released.

17:39:35 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":88}
17:39:35 RSL: stat/sonoff/POWER = ON
17:39:35 RSL: stat/sonoff/RESULT = {"POWER":"OFF","Dimmer":88}
17:39:35 RSL: stat/sonoff/POWER = OFF
17:39:37 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":88}
17:39:37 RSL: stat/sonoff/POWER = ON
17:39:38 RSL: stat/sonoff/RESULT = {"POWER":"OFF","Dimmer":88}
17:39:38 RSL: stat/sonoff/POWER = OFF
17:39:42 MQT: Attempting connection...
17:39:43 DNS: Query done. MQTT services found 0

digiblur, 2019-01-29:
I would make sure the switchtopics are set. You can break up the backlog commands and push each one in at a time.

33-viper-33, 2019-01-29:
just want to make sure that I understand.  So it should look like this?

33-viper-33, 2019-01-29:
That appears to have solved it.  Thanks for the help!  I'll install the switch and report back if it still isn't working.  Any idea why it wasn't getting properly set?  I was playing with generic device attempting to figure the dimmer out prior to stumbling across your code.  Is it possible it was storing previous settings from GPIO configuration?

digiblur, 2019-01-29:
Not sure as I do think when you change the module it erases a lot of settings. Plus I have seen it not take all the backlog commands sometimes.

Glad it worked out for you!

devanl, 2019-01-29:
Sorry to hijack the thread. I've been running this great with about 6
MJ-SD02 devices now so thank you very much for putting this together.  I
had a couple thoughts based on my experiences:
- could we create a saved setting that allows us to specify the min and max
duty cycle for the dimmer scale?  This way brightness always stays in the
effective range of LED bulbs (varies by manufacturer)
- I tried setting this up as a 3-way using the `dimmer` and `power` events
to publish to the other device cmnd topic and I'm running in to a problem
where rapid successive tapping of the button causes the devices to
ping-pong back and forth.
  - Any thoughts for a way to do this better?

33-viper-33, 2019-01-29:
Ok, just hooked it back to grid and there is still a problem. Pushing the power button ramps it from 0-100% brightness, COOL!.  Problem is the same thing happens when I momentarily press the up and down buttons.  It appears that all buttons are tied to the same function.

33-viper-33, 2019-01-29:
> Sorry to hijack the thread. I've been running this great with about 6 MJ-SD02 devices now so thank you very much for putting this together. I had a couple thoughts based on my experiences: - could [...]

No problem.  I'm actually looking to install 2 more non dimmable switches in a US 3-way setup and have this switch publish a long press to those switches.  So stairway light has switch at top and bottom of stairs.  Dimmable bedroom switch also activates stairway lignt with long press...  I'm curious for the answer too.  I saw some youtube videos of one of the guys doing a publish command in 3 way.  Would assume that the same would work for a fourth or fifth switch...

33-viper-33, 2019-01-29:
Disregard my last comment.  I reapplied your published rule1, and it started working as expected.  Thanks for the help.  I'm going to close this thread here.

closed 2019-01-29 by 33-viper-33

digiblur, 2019-01-29:
Awesome. I had it on my list and will take it off. I will open a new thread with the @devanl question.

digiblur, 2019-01-30:
@devanl couple things... I wonder if the min max could be set using some rules that check for value below or under; on Dimmer#State>95 do dimmer 95

I don't see anything off hand in the commands that allow a PWM min max value to be set.  

For the 3 way question, what are you trying to do with the 3 way?  You should be able to just send a power toggle to the dimmer from another device as this is what I do on a couple switches when using some long press on/off button actions I turn on some dimmers via MQTT.

devanl, 2019-01-30:
@digiblur I had worked out the rules approach before I came across your
fork and it was pretty slow. I don’t think that it would work with your
fork because the led indicator is driven directly from dimmer. What I'd
really like here would be an ability to set Max and min such that the
dimmer code applies this offset to the tele value (dimmer pwm = (dimmer
tele  * ((max-min)/100))+ dimmer min).  This way the range of the mqtt/tele
value is only within the operating range of lights connected.

With the 3-way I want to control power and dimming from two instances of
this device. I have rules that publish on dimmer and power events. In
general this works great. When someone impatient jams on the buttons
repeatedly I can run in to a race condition where the dimmer or power
states can oscillate between the units. One updates to new received power
and publishes based on rule event to original sender who is sending a new
value for next button press. The result is they continue to send each other
different values. I was thinking of making a rule that acts as a watchdog
to reset the device if this happens for too long but I'd rather it didn't
happen at all.