digiblur/Tasmota #2: Device will not go to AP mode if WIFI SSID is changed

ringmor, 2019-02-04:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : 
- [ ] _Development/Compiler/Upload tools used_ :
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ ] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
When for the first time device is connected to your WIFI all works, but when you change your WIFI password on your access point, device will not go to AP mode.

**Expected behavior**
When device can not connect to WIFI, it should go to AP mode in order to connect to internet

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
Original Tasmota firmware works as expected, but the modified version does not. Please not that all other functionalities work flawlessly as expected.

**(Please, remember to close the issue when the problem has been addressed)**

digiblur, 2019-02-04:
WiFi Manager isn't default as it follows the default configs in Tasmota so this is expected.  It's probably not a bad idea being these are switches to change the default to WIFI_MANAGER.  

Change your wificonfig option to 2 via the console and it will mimic this same behavior.

digiblur, 2019-02-13:
New builds have WIFI Manager set as default.  If you have a previous flash you'll need to set wificonfig to 2

closed 2019-02-13 by digiblur