digiblur/Tasmota #4: Moes 3 Way Switch Error

33-viper-33, 2019-03-23:
Hey Digiblure,

I'm having an issue with the Moes 3 Way Switch using you YouTube guide.  After pasting in your code, the physical switches work as advertised.  I then turned off MQTT and turned on Belkin WeMo to use directly with Alexa.  Alexa finds the switch without issue. When I attempt to issue an on/off voice command, it switches states of Toggle 1, but the light isn't effected.  To change states on the light, I must change states of Toggle 2.  Can you advise me how to fix the code?

Thanks,
-Viper

digiblur, 2019-03-24:
That's not going to work though as it comes up with two switches due to the way this switch is, it doesn't really have a true on/off.  You'd need to configure it with a home automation software.

digiblur, 2019-03-24:
You could try to swap the switches and button/switch configs around?  I haven't tested this.

33-viper-33, 2019-03-24:
I tried the switch and button swap.  It fails and locked up the switch.  The web interface locked up and nothing is responding.  I have to reflash to get it working again.  Specifically, I swapped 18 and 9.

digiblur, 2019-03-24:
It might have gone into factory reset mode if it was set as a button.

33-viper-33, 2019-03-24:
Had to reflash to get switch to respond.  I tried to swap the button and switch for two switches.  A second switch causes the relay/light to flash.  Also, the second toggle is always the the relay control.  I need to make the first toggle the relay control.  Not sure how to do that.

digiblur, 2019-03-24:
Did you swap the relays around too? I don't remember the exact reasoning I did that order.

33-viper-33, 2019-03-24:
That was dumb on my part.  I should have spotted that.  That fixes half the issue.  Alexa and the dumb switch now work.  When pressing the smart switch button, it changes the relay state only momentarily.  Tired switching the button to a switch, but that bricked it again.  Back to flashing...  I suspect a relay needs to be changed to the inverse...  Thoughts?

33-viper-33, 2019-03-24:
Well I feel stupid, that was half the problem and I should have guessed it.  Alexa now controls the relay and the dumb switch turns on and off the light.  Only problem now is with the smart switch button.  Pushing it momentarily changes state but reverts back immediately...  I'm going to experiment, but advice is welcome.

digiblur, 2019-03-24:
Try to change the switchmodes if it is toggling immediately back.

33-viper-33, 2019-03-24:
I tried SwitchMode 1 which keeps light on when not pressed and off while pressed.  Turns back on after release.  SwitchMode 2 is the exact opposite.

33-viper-33, 2019-03-24:
SwitchMode 3 fixed it!

33-viper-33, 2019-03-24:
All works when you swap the two relays, swap the two switches and change to SwitchMode 3.  Thanks for all your work and your support digiblur!

closed 2019-03-24 by 33-viper-33

digiblur, 2019-03-24:
Awesome glad you were able to figure it out!  I might try to swap mine around and see how this works, but I have them both deployed in the house in two locations and not on the bench.  Don't fix it if it isn't broke type thing might apply for me here.