digiblur/Tasmota #5: Boot loop when using firmware with Tekin plugs

DarrenRatcliffe, 2019-03-24:
When applying the latest firmware bin it puts tekin sp23 plugs (and other tekin plugs) in to a boot loop and I can not find a way to end the cycle to flash with other firmware.

[image]

digiblur, 2019-03-24:
Although it should work, I would use the regular 6.5.0 release of Tasmota as this is just for the Martin Jerry dimmer.

DarrenRatcliffe, 2019-03-24:
Ah OK!!, any idea how I can arrest the boot loop long enough to load new firmware?

digiblur, 2019-03-24:
hurry up and do a reset 5 while you can?  Did you enable the auto discovery?

DarrenRatcliffe, 2019-03-24:
reset 5 in the console? also which is the cmd to enable the device to default to an access point if there is a problem with wifi? thanks

DarrenRatcliffe, 2019-03-24:
if by setting SetOption19 1 then yes i enable auto discovery

digiblur, 2019-03-24:
Yes. There is a bug in Tasmota of the version I had that I updated the dimmer to.

Turn off setoption19 and it will stop rebooting. 

And always set wificonfig 2 btw.

closed 2019-03-27 by digiblur