digiblur/Tasmota #7: software works... dimmer on/off switch doesn't

helgew, 2019-04-08:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used :
- [x] Hardware used : _MartinJerry MJ-SD01_
- [ ] If a pre-compiled release or development binary was used, which one? :
- [x] You have tried latest release or development binaries? : _Yes. Compiled from today's master branch._
- [x] Provide the output of command ``status 0`` :

```
{"Status":{"Module":70,"FriendlyName":["DiningDimmer"],"Topic":"sonoff/mj-diner","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[6,5,5,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
```
---

This most certainly is a hardware issue as the Web GUI works as it should. That said, I am hoping to find some help with troubleshooting here.

Basically, the dimmer's on/off button only works when the low voltage board and the switch plate part (connected by the four little screws; the "lid") are not inserted into the cup-like high voltage/relay portion of the dimmer (the "cup"). Sometimes, when holding the lid close to the cup and/or attaching the ground plate to the lid, the on/off switch also stops working.

The reset switch and the up and down switches work as they should, although long-pressing the down switch only gets me down to 20%.

Any help would be greatly appreciated. Thanks for all your work on this.

digiblur, 2019-04-08:
Did all of your rules and switchmodes complete?

helgew, 2019-04-08:
they did! at least as far as I can tell. - h.

digiblur, 2019-04-08:
Make sure all the rules are in and turned on by typing Rule1, Rule2, etc commands at the console. Otherwise indeed it does sound like a hardware failure.

helgew, 2019-04-08:
I only have Rule1 set according to your instructions. As I mentioned, all the logic works either via GUI or on HomeAssistant/HomeKit. I double-checked the solder points I used but they seem fine.

Ordered another one to swap out just the lid and see what happens.

helgew, 2019-04-10:
I just got the replacement dimmer and it works as it should. Closing this as it does seem to be a hardware issue.

closed 2019-04-10 by helgew

digiblur, 2019-04-10:
Glad you figured it out! Thanks for the update.