digiblur/Tasmota #8: Treatlife dimmer flashing issue

Mtbike, 2019-04-17:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : used pre-compiled 
- [x] Hardware used : Treatlife dimmer
- [x] If a pre-compiled release or development binary was used, which one? : 	Tasmota_6.5.0(digiDIMv6).bin,  	Tasmota_6.5.0(digiDIMv5).bin and sonoff 6.5
- [x] You have tried latest release or development binaries? :yes
- [x] Provide the output of command ``status 0`` :
```
I have been trying to flash a Treatlife dimmer first with Tuya-Convert and now by soldering. I am able to flash it with FlashESP8266.exe but when i connect to it using termite I the following. 

ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3fff3378
~ld
system param error
fc e7 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff 
fc e7 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
[... 11 lines cut ...]
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff æâÜ[00]Œœ[1C][02]Ä[1C]âÜ[00]Œœ[1C][02]Ä[1C]âÜ[00]Œœ[1C][02]Ä[1C]âÜ[00]Œœ[1C][02]Ä;ˆ[02][1C]âÜ[00]Œœ[1C][02]Ä[1C]âÜ[00]Œœ[1C][02]„;[08][02]€l[1C][00]Œœ[1C][02]Ä[1C]âÜ[00]Œœ[1C][02]Ä[03][12]Ü[00]lŽÛ[0C][18][02][12]Ü[00]lŽÛ[0C][18]âŒ[1C][00][0C][1C]€[0C][18]€[0C][1C][00][0C][1C]€[0C][18]àì8[00][0C][1C][0C][18]€[0C][1C][00][0C][1C]€[0C][18]àì8[00][0C][1C][0C][18]€[0C][1C][00][0C][1C]€[0C][18][1C]‚Ç[00][0C][1C]€[0C][18]€[0C][1C][00][0C][1C]€[0C][18]âŒœ[00][0C][1C]€[0C][18]€[0C][1C][00][0C][1C]€[0C][18]r[18][02]

```

**(Please, remember to close the issue when the problem has been addressed)**

digiblur, 2019-04-17:
I had a similar issue with it rebooting via USB. The secondary MCU requires more power and I had to power it via another voltage regulator on the same USB bus. I used a nodemcu and it worked.

Mtbike, 2019-04-17:
Could you explain the wiring between the FTDI232, NodeMCU and Treatlife chip?

digiblur, 2019-04-17:
You can see I had the issue in my livestream of it...think it was day 2.  It kept rebooting as even my little better CH340G usb adapter wouldn't power it.  So in a pinch I decided let's see what happens.  I had two USB ports on the same LCD screen to have a common ground.  I took a NodeMCU powered via USB cable and went off the 3v3 and gnd pins on the NodeMCU.  Then I attached the RX/TX from the dimmer to the USB flasher along with attaching its ground to the same ground the nodemcu had.  It was then able to power the unit without it constantly rebooting.  

Did Tuya Convert not work?  I just did this one right after the livestream without issue.

Mtbike, 2019-04-17:
I could not get tuya convert to work. The issue was the switch kept getting the ip address for 10.42.42.31. I even tried changing the ip address in the script from 10.42.42.42 to .31. This allowed it get pass the step of just displaying .............. But I could not get passed that point and just soldered it.

Mtbike, 2019-04-17:
I am watching your livestream and noticed I soldered wrong rx/tx pins. I am going to try the other set. Not sure what issue I caused using flashing tasmota on the U0RX/U0TX

digiblur, 2019-04-17:
You will need the other set of RX TX..not the ones I used. I was trying to get the Tuya packets.

Mtbike, 2019-04-17:
I got the extra power hooked up and working. But after flashing
Power the unit down
Disconnect GPIO 0 from ground
Power it back on 
Start termite
This is what is fills up termite 
[1B][02]ß[07]’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€l`[1B]Ä[03]‚ß[03]‚nÀ$œàrÜ[1B][02]ßØ[12]nnß[02]Ä [0C][1C]bŒ[1F][00]l`[1B]$œÀrÜÃl[0C][1C]à‚ÇÀl[1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÄl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÇl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œàrÜÜ[12]Ü[18]‚œÄ’[0C][1C]à[02]‡À[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ[04]œ[1B][02]ß[07]’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€l`[1B]Ä[03]‚Ÿ[03]‚nÀ$œàrÜ[1B][02]ßØ[12]nnŸ[02]Ä [0C][1C]bŒ[1F][00]l`[1B]$œ€rÜÃl[0C][1C]à‚ÇÀl[1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÄl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÇl[0C][1C]à[02]‡À[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œàrÜÜ[12]Ü[18]‚ÜÄ’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ[04]œ[1B][02]Ÿ[07]’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€l`[1B]Ä[03]‚ß[03]‚nÀ$œàrÜ[1B][02]ßØ[12]nnß[02]Ä [0C][1C]bŒ[1F][00]l`[1B]$œÀrÜÃl[0C][1C]à‚ÇÀl[1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÄl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÇl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œàrÜÜ[12]Ü[18]‚ÜÄ’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ[04]œ[1B][02]ß[07]’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€l`[1B]Ä[03]‚ß[03]‚nÀ$œàrÜ[1B][02]ßØ[12]nnß[02]Ä [0C][1C]bŒ[1F][00]l`[1B]$œÀrÜÃl[0C][1C]à‚ÇÀl[1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÄl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrœÇl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]b€$œàrÜÜ[12]Ü[18]‚ÜÄ’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ[04]œ[1B][02]ß[07]’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€l`[1B]Ä[03]‚ß[03]‚nÀ$œàrÜ[1B][02]ßØ[12]nnß[02]Ä [0C][1C]bŒ[1F][00]l`[1B]$œÀrÜÃl[0C][1C]à‚ÇÀl[1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÄl[0C][1C]à[02]‡À[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÇl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œàrœÜ[12]Ü[18]‚ÜÄ’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ[04]œ[1B][02]ß[07]’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€l`[1B]Ä[03]‚ß[03]‚nÀ$œàrÜ[1B][02]ŸØ[12]nnß[02]„ [0C][1C]bŒ[1F][00]l`[1B]$œÀrÜÃl[0C][1C]à‚‡Àl[1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÄl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÇl[0C][1C]à[02]‡À[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œàrÜÜ[12]Ü[18]‚ÜÄ’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ[04]œ[1B][02]ß[07]’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€l`[1B]Ä[03]‚ß[03]‚nÀ$œàrÜ[1B][02]ßØ[12]nnß[02]Ä [0C][1C]bŒ[1F][00]l`[1B]$œÀrÜÃl[0C][1C]à‚ÇÀl[1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrœÄl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÇl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œàrÜÜ[12]Ü[18]‚ÜÄ’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ[04]œ[1B][02]ß[07]’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€l`[1B]Ä[03]‚ß[03]‚nÀ$œàrÜ[1B][02]ßØ[12]nnß[02]Ä [0C][1C]bŒ[1F][00]l`[1B]$œÀrÜÃl[0C][1C]à‚ÇÀl[1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÄl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÇl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œàrÜÜ[12]Ü[18]‚ÜÄ’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ[04]œ[1B][02]ß[07]’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€l`[1B]Ä[03]‚ß[03]‚nÀ$œàrÜ[1B][02]ßØ[12]nnß[02]Ä [0C][1C]bŒ[1F][00]l`[1B]$œÀrÜÃl[0C][1C]à‚ÇÀl[1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÄl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÇl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œàrÜÜ[12]Ü[18]‚ÜÄ’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ[04]œ[1B][02]ß[07]’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€l`[1B]Ä[03]‚ß[03]‚nÀ$œàrÜ[1B][02]ßØ[12]nnß[02]„ [0C][1C]bŒ[1F][00]l`[1B]$œÀrÜÃl[0C][1C]à‚ÇÀl[1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÄl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÇl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œàrÜÜ[12]Ü[18]‚ÜÄ’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ[04]œ[1B][02]ß[07]’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€l`[1B]Ä[03]‚ß[03]‚nÀ$œàrÜ[1B][02]ßØ[12]nnß[02]Ä [0C][1C]bŒ[1F][00]l`[1B]$œÀrÜÃl[0C][1C]à‚ÇÀl[1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÄl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]b€$œÀrÜÇl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œàrÜÜ[12]Ü[18]‚œÄ’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ[04]œ[1B][02]ß[07]’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€l`[1B]Ä[03]‚ß[03]‚nÀ$œàrÜ[1B][02]ŸØ[12]nnß[02]„ [0C][1C]bŒ[1F][00]l`[1B]$œÀrÜÃl[0C][1C]à‚ÇÀl[1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÄl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÇl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œàrÜÜ[12]Ü[18]‚ÜÄ’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ[04]œ[1B][02]ß[07]’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€l`[1B]Ä[03]‚ß[03]‚nÀ$œàrÜ[1B][02]ßØ[12]nnß[02]„ [0C][1C]bŒ[1F][00]l`[1B]$œÀrÜÃl[0C][1C]à‚ÇÀl[1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÄl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÇl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œàrÜÜ[12]Ü[18]‚ÜÄ’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ[04]œ[1B][02]ß[07]’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€l`[1B]Ä[03]‚ß[03]‚nÀ$œàrÜ[1B][02]ßØ[12]nnß[02]Ä [0C][1C]bŒ[1F][00]l`[1B]$œÀrœÃl[0C][1C]à‚‡Àl[1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrÜÄl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrœÇl[0C][1C]à[02]‡À[0C][1C]€[0C][1C]€[0C][1C]€[0C]b€$œàrÜÜ[12]Ü[18]‚ÜÄ’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ[04]œ[1B][02]ß[07]’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€l`[1B]Ä[03]‚Ÿ[03]‚nÀ$œàrœ[1B][02]ßØ[12]nnß[02]Ä [0C][1C]bŒ[1F][00]l`[1B]$œÀrÜÃl[0C][1C]à‚ÇÀl[1C]€[0C][1C]€[0C][1C]€[0C]b€$œÀrœÄl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œÀrœÇl[0C][1C]à[02]‡À[0C][1C]€[0C][1C]€[0C][1C [... cut ...] ][1C]€[0C]bÀ$œÀrÜÄl[0C][1C]à[02]‡À[0C][1C]€[0C][1C]€[0C][1C]€[0C]b€$œÀrÜÇl[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ$œàrÜÜ[12]Ü[18]‚œÄ’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€[0C]bÀ[04]œ[1B][02]ß[07]’[0C][1C]à[02]ÇÀ[0C][1C]€[0C][1C]€[0C][1C]€l`[1B]Ä[03]‚ß[03]‚nÀ$œàrÜ[1B][02]ßØ[12]nnß[02]Ä [0C][1C]bŒ[1F][00]l`[1B]$œÀrÜÃl[0C]

I have also tried power it back up with GPIO 0 connected to ground and starting termite but when I do this I get no response when I use backlog command.

Any ideas?

digiblur, 2019-04-17:
I would probably just configure it over web since it has the serial comms although one would think it would be fine over GPIO1 and 3.

Mtbike, 2019-04-17:
Take that back I am able to connect to it now. I connected to my router and seen the new IP address

I was not able to use termite to set the wifi settings. I compiled a new version with ssid and password.

digiblur, 2019-04-17:
WiFi manager as default is good thing with these serial based dimmers.

Mtbike, 2019-04-17:
I am connected to the web interface but when I toggle it I do not have the relay toggle like it does when the button is pressed. Here is what I am seeing in the console log
22:50:27 TYA: Set WiFi LED to state 3 (0)
22:50:27 TYA: TX Packet: "55aa000300010306"
22:50:28 TYA: Set WiFi LED to state 3 (0)
22:50:28 TYA: TX Packet: "55aa000300010306"
22:50:29 TYA: Set WiFi LED to state 3 (0)
22:50:29 TYA: TX Packet: "55aa000300010306"
22:50:30 TYA: Set WiFi LED to state 3 (0)
22:50:30 TYA: TX Packet: "55aa000300010306"
22:50:31 TYA: Set WiFi LED to state 3 (0)
22:50:31 TYA: TX Packet: "55aa000300010306"
22:50:32 TYA: Set WiFi LED to state 3 (0)

I have it configured as a Tuya Dimmer. Thanks for the help

digiblur, 2019-04-17:
Did you set the RX and TX pins to gpio13 and 15 and set the dimmer ID?

Mtbike, 2019-04-17:
I did not, but i set them to what it said in your comments from the YouTube video. Now ever 10 seconds I see a new command (last line)
23:23:31 TYA: Set WiFi LED to state 3 (0)
23:23:31 TYA: TX Packet: "55aa000300010306"
23:23:31 TYA: TX Packet: "55aa00000000ff"

digiblur, 2019-04-17:
Yep. That is the heartbeat command it needs every 10 seconds. Is the dimming and on off working now?

Mtbike, 2019-04-17:
no, it does not change nor does it get updated when I do it manually

digiblur, 2019-04-17:
Can you post screenshots of your module config?

Mtbike, 2019-04-17:
I had the RX and TX backwards. Sorry. It is working now.

closed 2019-04-17 by Mtbike

digiblur, 2019-04-17:
Yay! Glad you got it!