digiblur/Tasmota #9: Home Assistant not accepting the brightness yaml.

mattmurphey25, 2019-05-18:
Home Assistant release with the issue:
0.92.2

Operating environment (Hass.io/Docker/Windows/etc.):
Hass.io, Docker

I followed the tutorial to a tee and everything works as it should, but I can't dim the bulb.
"Your configuration contains extra keys that the platform does not support.
Please remove [brightness_state_topic], [brightness_command_topic], [brightness_scale], [brightness_value_template].  (See /config/configuration.yaml, line 29)."

- [ ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : 
- [ ] _Development/Compiler/Upload tools used_ :
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ ] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

digiblur, 2019-05-18:
Make sure you put this under your "light" section and not "switch" as Home Assistant doesn't accept dimming on switches.

mattmurphey25, 2019-05-18:
That was it. Your tutorial is exceptional.

closed 2019-05-18 by mattmurphey25

digiblur, 2019-05-18:
Thanks for the update and comments!