digiblur/Tasmota #10: Treatlife Dimmer With TYWE3S Not Flashing

Mtbike, 2019-07-26:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used :
- [x] Hardware used : Treatlife Dimmer with TYWE3S
- [x] If a pre-compiled release or development binary was used, which one? : Tasmota_6.5.0(digiDIMv6)
- [x] You have tried latest release or development binaries? : No, will not go in to flash mode
- [x] Provide the output of command ``status 0`` :
```
 will not go in to flash mode
```
I am trying to flash it using the soldering method, I am not able to get it to flash. But I also purchased the new non dimming switch which also has the TYWE3S chip. The differen
![Dimmer with TYWE3S]
ce is that non dimming switch does not have the Rx and Tx, Rst or GPIO15 contacts soldered. Any help would be great. 

(https://user-images.githubusercontent.com/48741415/61955159-0cfb9600-af80-11e9-86bf-768876783794.jpg)

**(Please, remember to close the issue when the problem has been addressed)**

Mtbike, 2019-07-26:
I bought 30 dimming switches and 26 of them had the TYWE1S chip and I was able to flash all of them. The last 4 have the TYWE3S. I did not try Tuya Convert as I do not have it set up and from what I have read the newest treatlife switches no longer work with Tuya Convert.

digiblur, 2019-07-26:
I have seen a few times that some USB flashers on the dimmers do not have enough power to power the chips on the dimmers. I ran into this issue myself on the live stream trying to reverse engineer it.

Mtbike, 2019-07-26:
I had that problem with the TYWE1S chip. But I was still able to flash it with USB power.  I can try another power source.

FireAcidAP, 2019-07-27:
I am also in the same boat. I bought the 4 pack from amazon [Treatlife Dimmer Model: DS01C](https://www.amazon.com/Treatlife-Incandescent-Compatible-Assistant-Single-Pole/dp/B07PJTLB7Z?ref_=bl_dp_s_web_18705530011). Inside the dimmers are the TYWE3S chips. Switches come with firmware WiFi Module version 3.1.1 & MCU Module version 4.0.1. Tried Flashing Tuya-Convert from 2 different devices and nothing happens when the dimmer supposed to join vtrust-flash, Tried the NodeMCU PyFlasher with GPIO0 and Ground bridged when connecting FTDI without any luck. The device doesn't even seem to go into the boot mode. I am suspecting it has something to do with the MCU(BYD BF7612AM28T) which is connected to the TYWE3S over the TX & RX pins. I couldn't find any datasheets on this MCU so not sure on how to disable it from booting every time I apply power to the board... I am trying to stay away from de-soldering the TYWE3S from the board.
[image]
[image]
[image]
[image]
Hopefully this helps in devising a solution to resolve flashing issues for Treatlife dimmers. For now stuck in using the SmartLife app.

digiblur, 2019-07-28:
@FireAcidAP what have you tried powering it up with?  That always seems to be the issue with these is getting enough power to the device to take flash.  

I'm thinking about ordering one of these myself and using a separate DC power supply for cases like this.  https://www.crowdsupply.com/pylo/muart

Mtbike, 2019-07-28:
I tied a NodeMCU and it had a white light that would fade in and out. It would not flash, how do you tell if it is in flashing mode?

digiblur, 2019-07-28:
The white light fade in and out is normal on the one I have installed. Hard to tell if it is not in flash mode without it hooked to mains. Do you have it paired with Smartlife? If so if it pops up as online then you know it isn't in flash mode.

Mtbike, 2019-07-29:
On a side note I have new treatlife Non dimming switch. I flashed it with Tasmota and looking for suggestions on what type of module I should select. It has the same white  and red LEDs at the bottom as the dimming one.

Here is what I know

Red LED - gpio 5
White LED - gpio 4
Button NO contact gpio 13
Button Common - Gnd 

And help or pointing in the right direction would be great

FireAcidAP, 2019-07-30:
> @FireAcidAP what have you tried powering it up with? That always seems to be the issue with these is getting enough power to the device to take flash.
> 
> I'm thinking about ordering one of these myself and using a separate DC power supply for cases like this. https://www.crowdsupply.com/pylo/muart

Thanks for the reply @digiblur. I have tried with using the FTDI's power only, additional [Soldering Station Power supply](https://www.amazon.com/F2C-853d-Supply-Soldering-Station/dp/B018LFTWTK/ref=asc_df_B018LFTWTK/?tag=hyprod-20&linkCode=df0&hvadid=312128028232&hvpos=1o3&hvnetw=g&hvrand=14569226333991911601&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9008145&hvtargid=pla-571443016695&psc=1) only and combination of both (without the +3.3v from FTDI). 
I am now able to flash the Tasmota firmware to the TreatLife dimmers DS01C with TYWE3S chip. I had few issues as follows.
**Problem 1** I was applying 3.3v to the TYWE3S chip instead of the the header pins on the board. If 3.3v was applied to the chip directly the voltage would be dropped to 0.9v at the TYWE3s chip. 
**Problem 2** I weren't able to get TYWE3S chip into the boot mode. For that I had to wire up the chip with minimal circuit requirements as per this [instructable](https://www.instructables.com/id/ESP-12F-ESP8266-Module-Minimal-Breadboard-for-Flas/). Once the chip was in the boot mode the flashing still resulted in _espcomm_sync failed_, For reference, the boot mode can be verified through Serial Monitor @ 74880 baud the line will say something like _ets Jan  8 2013,rst cause:2, boot mode: (1, 6)_, 
**Problem 3** was the aforementioned **espcomm_sync failed** error in Arduino IDE and NodeMCU PyFlasher would just not start flashing. That I had to resolve by cutting the RX trace from the TYWE3S chip to the MCU. Once i finished with flashing the Tasmota Firmware i have soldered the cut RX trace back together. Powered the unit on and tested to make sure everything is working properly in the new firmware before putting everything back together.
First Dimmer I cut both traces just in case :)
[image]
Second Dimmer i cut only RX and the flashing worked.
[image]
Hopefully this information can be helpful to somebody else.

digiblur, 2019-08-03:
Thanks for the great info! Sounds a lot like the Costco dimmer mod to get those to flash.

closed 2019-08-05 by Mtbike