digiblur/Tasmota #11: Flashed on MOES  DS01

vajonam, 2019-09-04:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used :
- [x] Hardware used :  MOES DS01 - TYWE1S based device 
- [x] If a pre-compiled release or development binary was used, which one? : Tasmota_6.6.0.6(digiDIMv7).bin
- [x] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :

```
00:03:00 RSL: stat/sonoff/STATUS = {"Status":{"Module":54,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
00:03:00 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:00","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":10,"SaveCount":62,"SaveAddress":"F6000"}}
00:03:00 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0.6(digiDIMv7)","BuildDateTime":"2019-08-25T13:37:05","Boot":7,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:03:00 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["gaul-hw",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008029","280503000100060000005A00000000000000","00000000"]}}
00:03:00 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":448,"Free":552,"Heap":20,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":3,"Features":["00000809","0F402392","00168000","00000014","000000C0","00000000"],"Drivers":"1,2,3,4,6,9,10,12,16,18,19,20,21","Sensors":"2,3,4,5"}}
00:03:00 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3506","IPAddress":"10.1.1.7","Gateway":"10.1.1.253","Subnetmask":"255.255.255.0","DNSServer":"10.1.1.253","Mac":"B4:E6:2D:66:AD:B2","Webserver":2,"WifiConfig":2}}
00:03:00 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_66ADB2","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
00:03:00 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:03:00 1970","Local":"Thu Jan 01 00:03:00 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
00:03:00 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:03:00","Epoch":180}}
00:03:00 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:03:00","Epoch":180,"Uptime":"0T00:03:00","UptimeSec":180,"Heap":20,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":455,"MqttCount":0,"POWER":"ON","Dimmer":6,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"gaul-hw","BSSId":"92:2A:A8:1A:31:3D","Channel":11,"RSSI":80,"LinkCount":1,"Downtime":"0T00:00:04"}}}
00:03:01 TYA: Set WiFi LED 3 (0)
00:03:01 TYA: Send "55aa000300010306"
00:03:02 TYA: Set WiFi LED 3 (0)
00:03:02 TYA: Send "55aa000300010306"
00:03:03 TYA: Set WiFi LED 3 (0)
```

So I have issue where I think the MCU isn't being properly set. I had to set GPIO 13 and 15 to RX and TX and it starts working, but once I do a reboot. Nothing works, lights on the front disappear, relay stops clicking.. not sure what to do get it back working.. I have tried 2 switches same issue. The web / console and connection to mqtt work fine.

vajonam, 2019-09-04:
```

00:00:20 LOG: SerialLog 0, WebLog 4, SysLog 0, LogHost , LogPort 514, TelePeriod 300
00:00:20 HTP: Configuration
00:00:21 TYA: Set WiFi LED 3 (0)
00:00:21 TYA: Send "55aa000300010306"
00:00:21 CFG: Saved to flash at F8, Count 12, Bytes 3584
00:00:22 HTP: Main Menu
00:00:22 TYA: Set WiFi LED 3 (0)
00:00:22 TYA: Send "55aa000300010306"
00:00:23 HTP: Console
00:00:23 TYA: Set WiFi LED 3 (0)
00:00:23 TYA: Send "55aa000300010306"
00:00:23 TYA: Send "55aa00000000ff"
00:00:24 TYA: Set WiFi LED 3 (0)
00:00:24 TYA: Send "55aa000300010306"
00:00:25 TYA: Set WiFi LED 3 (0)
00:00:25 TYA: Send "55aa000300010306"
00:00:26 TYA: Set WiFi LED 3 (0)
00:00:26 TYA: Send "55aa000300010306"
00:00:27 WIF: Checking connection...
00:00:27 WIF: Connected
00:00:27 TYA: Set WiFi LED 3 (0)
00:00:27 TYA: Send "55aa000300010306"
```

digiblur, 2019-09-24:
Not sure why I didn't get emailed this issue...  did you figure it out?

vajonam, 2019-09-25:
Nope ended up giving up..  we tried some newer mainline builds. I do have one un patched switch remaining.. will try it later this weekend, any fixes that might fix this issue been made since the last couple of weeks?

digiblur, 2019-09-25:
Do you have a link to this device?

vajonam, 2019-09-25:
[amazon canada](https://www.amazon.ca/MOES-Control-Compatible-Required-Standard/dp/B07PLCMR71/ref=sr_1_1_sspa?keywords=moes+dimmer&qid=1569444366&s=gateway&sr=8-1-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUFQTFZGVEVZNFdMN0omZW5jcnlwdGVkSWQ9QTAzODA4ODgyN1hXVFlaT0U4WVomZW5jcnlwdGVkQWRJZD1BMDM3NzYxMjNUS1dUWDU2U09KQ0cmd2lkZ2V0TmFtZT1zcF9hdGYmYWN0aW9uPWNsaWNrUmVkaXJlY3QmZG9Ob3RMb2dDbGljaz10cnVl)

digiblur, 2019-09-25:
Ahh.. I have one of those in my kitchen. What version of firmware are you using?

vajonam, 2019-09-26:
Last I tired both 6.6.0.10 and 6.6.0.11. I for whatever reason i looks like
the MCU is resetting or losing power. but I tried 3 different switches the
same happens to all of them. Not sure why. I have been soldering wires on
to flash. I might try just holding some pins on there. to see if my
soldering is fouling things up..

digiblur, 2019-09-26:
Very strange. You could try some older versions just to be sure as well as using the newer updated Tuya Convert process to take out the soldering.

vajonam, 2019-09-26:
Yup. That’s what I am doing now. Just getting my rbp ready for Tuya covert
v2

vajonam, 2019-09-26:
Okay! TuyaConvert V2 worked a charm, installed 6.5 to start and upgraded to
6.6.0.14 everything seems to be work as per the wiki after TuyaMCU 21,3.
Have it install let me keep an eye on it as I get few more switches!

digiblur, 2019-09-26:
Awesome glad you got it working!

closed 2019-09-26 by digiblur