digiblur/Tasmota #15: Tasmota_7.0.0.5(digiDIMv8).bin missing Wemo/Hue Emulation

muchtall, 2019-12-02:
It seems that the Tasmota_7.0.0.5(digiDIMv8).bin image does not contain the Wemo and Hue emulation support. The configuration options are missing from the Configure Other page, and the "Emulation" command is unknown.

qosmio, 2019-12-02:
Can confirm also facing this issue. Had to revert to 6.6.0.6

digiblur, 2019-12-03:
I will add those back and hopefully it won't be too big.

digiblur, 2019-12-04:
Look in this folder for the alternate one with hue, wemo and domo..  https://github.com/digiblur/Tasmota/tree/development/bins

@qosmio @muchtall

closed 2019-12-04 by digiblur

ToddNJ, 2020-02-02:
@digiblur did you overlook adding these into the newest version, or was the 8.x version too fat with the emulation?
Thanks!