digiblur/Tasmota #17: Compile Error with PlatformIO

MTGS69, 2020-02-23:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ X] Searched the problem in the chat (https://discord.gg/cPAxGqm)
- [ X] Development/Compiler/Upload tools used : VSCODE (with latest PLATFORMIO
- [ ] Hardware used : Windows 10
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] Provide the output of command ``status 0`` : N/A
```
STATUS 0 OUTPUT HERE
```
I've always compiled my own builds in PLATFORMIO.  Since the release of 8.0 I started having build problems.  I finally did a complete reinstall of VSCODE, deleted all the associated directories, reinstalled the PLATFORMIO extension and finally had success with the development build of 8.01.0.9.  I am now trying to build your fork for use with Martin Jerry Dimmer.  I downloaded the latest Development branch.....changed nothing as a test....and get the same build error I was getting with the new 8.0 version before starting from scratch.  Specifically......

Can not remove temporary directory `C:\My Program Files\PLATFORMIO\SOURCE\Tasmota-development\.pioenvs`. Please remove it manually to avoid build issues
Error: Invalid 'c:\My Program Files\PLATFORMIO\SOURCE\Tasmota-development\platformio.ini' (project configuration file): 'No option 'platform_packages' in section: 'common''
The terminal process terminated with exit code: 1

Again....as a test I did not change anything in the source files.  I also was sure to perform and Update All from within PLATFORMIO so I am using the latest version of everything.  Any help would be greatly appreciated.  I can install your precompiled bins (work great.....thanks) but there are a few changes I would like to make to support my standard flash process (preset options within my_user_config.h).

Kind Regards.....Mike

**(Please, remember to close the issue when the problem has been addressed)**

OriginalHotrodder, 2020-03-05:
I have the exact same issue as you Mike.  When I build the Development source files, all works great.  Hoping for a fix.
Best regards,  Edmund

digiblur, 2020-03-05:
I know I copied the files over that I used but I probably won't be using this fork anymore if all tests go well with the new pwm dimmer template being built in and work fine with my alternate rules.

MTGS69, 2020-03-07:
Greatly appreciate the reply.  I noticed that the latest dev build has support for the MJ-SD01 (PWM) dimmers.  I successfully compiled it with my typical custom settings (wifi info etc.).  However, once I changed the module over to "PWM Dimmer" none of the on-device LED lights worked like your previous build (have not even installed it into the wall to see if the buttons at least function as expected).  I also tried implementing your 3 rules hoping that would be fix....but to no avail.  Please let me know if you figure that part out and I shall do the same.  Again....thanks for the work you do....I enjoy your YouTube vids.  Keep up the great work...Mike

digiblur, 2020-03-08:
Testing it on one and the status lights come up fine and I made a template with switches to do some rules but switchmode 6 seems to be broken in the latest version. Had to rolll back a commit for that.

MTGS69, 2020-03-08:
The status lights work fine but, at least with the default config a) no MQTT message is sent when the dim level is changed via the physical switch and b) I cannot figure out how to enable HOLD.....which is probably related to your comment above.  

As an aside/FYI, I tried toggling on SetOption88 just to play with it.  Once set, if I press a physical button the device reboots.  Tx!!

digiblur, 2020-03-08:
I saw Theo rolled back switchmode 6 so it should work now using my old school rules, set the template to switch1, 2, 3 and roll on.

MTGS69, 2020-03-08:
Sweet!  Appreciate the help!!!

MTGS69, 2020-03-08:
Hmmmm....still no work.  Perhaps I am missing something stupid.  Downloaded the latest build from the main repo......changed my_user_config.h for my preferred settings (see attached if it helps)....and successfully compiled via platformIO.   Performed an erase flash, uploaded the new bin, changed then Module type to PWM Dimmer (73).  Lastly, I copied your first rule (the one that processes switch presses and sets the switch modes to 5/6).  Regardless if the rule is set to on or off I get the same behavior:  1) press of dimmer up or down does nothing 2) hold of dimmer up down adjusts the dim level but no MQTT messages appear in the web terminal (but even if they did I cannot assign custom actions to HOLD like your Rule2).  Is there anything else I need to do with the main branch version to make it work like your fork?  Note: The dimmer LEDs are/have been working fine out of the box now.
[my_user_config.zip](https://github.com/digiblur/Tasmota/files/4304010/my_user_config.zip)

digiblur, 2020-03-08:
Jumping ahead of instructions I am writing up. :) set your template to switches and activate it?

digiblur, 2020-03-10:
@MTGS69 try this template with the rules.  Don't forget to enable it.  

{"NAME":"digiDIM","GPIO":[11,10,0,59,158,58,0,0,57,37,56,9,29],"FLAG":0,"BASE":73}

MTGS69, 2020-03-11:
It verks!  One inconsequential difference between your fork and the new master...Rule 3 does not work.  I suspect because control of the red-blue power led is via the firmware.  My preference would be for the blue led to be on when power is off and never use red (so you can find it in the dark without being too obtuse on the eyes)...but no big deal.  I just signed up for your $2/mo tier Patreon account....the least I can do for saving me from having to design my own dimmer switch....too many projects....not enuf time. Thanks again for the help!!!

MTGS69, 2020-03-11:
Officially closing this issue as problem was solved.

closed 2020-03-11 by MTGS69

digiblur, 2020-03-11:
Awesome! The blue can't be changed unfortunately only the red as the blue is hardwired . The rule 3 isn't needed anymore as the setoption87 will do that now.

MTGS69, 2020-03-11:
@digiblur Holy $#@!.....up to 89 SetOptions now.  It amazing how much Theo crams into Tasmota and why I gave up wasting time programming my own 8266 firmware years ago.  Well if blue cannot be changed to off (ah ha...one of those 2 color led with 3 legs)  SetOption87 will remain off.  Red leds in the dark score low on the wife acceptance factor scale and give me nightmares:)

MTGS69, 2020-03-11:
I dug deeper and removed the four screws attaching the faceplate to the daughter board.....nothing a soldering iron/swap cant fix!