digiblur/Tasmota #18: Dimm level bellow 10% ?

raufis27, 2020-02-27:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the chat (https://discord.gg/cPAxGqm)
- [ ] Development/Compiler/Upload tools used :
- [ x] Hardware used : 
- [x ] If a pre-compiled release or development binary was used, which one? : Tasmota_7.0.0.5(digiDIMv8).bin
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
05:03:35 MQT: /myhome/smart_light/stat/kitchen_dimmer/STATUS = {"Status":{"Module":72,"FriendlyName":["Kitchen Dimmer"],"Topic":"kitchen_dimmer","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":8,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[6,5,5,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
05:03:35 MQT: /myhome/smart_light/stat/kitchen_dimmer/STATUS = {"Status":{"Module":72,"FriendlyName":["Kitchen Dimmer"],"Topic":"kitchen_dimmer","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":8,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[6,5,5,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
05:03:35 MQT: /myhome/smart_light/stat/kitchen_dimmer/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Software/System restart","Uptime":"0T00:15:46","StartupUTC":"2020-02-27T03:47:49","Sleep":50,"CfgHolder":4617,"BootCount":48,"SaveCount":13734,"SaveAddress":"F8000"}}

```
First of all thank you so much for your hard work. It is super easy to configure dimmers with your version.
I have question about dimming level:
So I am using same LED bulbs with MJ dimmer and ZWave switch. With Zwave I can dimm as low as 5%, but with MJ and Tasmota I can't go bellow 10%. Is there any way to fix this ?

**(Please, remember to close the issue when the problem has been addressed)**

digiblur, 2020-02-27:
That's all based on the bulbs and dimmer combinations actually.  On some bulbs I can do down to 1% on the MJ dimmer, and then some other bulbs I can't get below 15 to 20%.

raufis27, 2020-02-27:
Ok, I guess Zwave dimmer is smarter. I just found this page [https://github.com/arendst/Tasmota/wiki/tuyamcu-configurations](url)

raufis27, 2020-02-27:
Could you please share link to bulbs with max dimming compatibility or model Information ?

digiblur, 2020-02-27:
Different internal dimming designs. In my opinion a zwave dimmer is a just dumb remote controlled switch.

closed 2020-02-27 by digiblur