digiblur/Tasmota #20: Dimmer Minimum level

raufis27, 2020-04-18:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the chat (https://discord.gg/cPAxGqm)
- [x ] Development/Compiler/Upload tools used :
- [x ] Hardware used : MJ-SD01 Dimmer 
- [x ] If a pre-compiled release or development binary was used, which one? : Tasmota_8.2.0.1(digiDIMv10).bin
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**
Upgrading to the 8.2.0.1 from 8.1.0.1 results in minimum effective dimming level being between 40-50%. Everything works as expected with previous release. 10% minimum 100% maximum dimming level. Any ideas.

raufis27, 2020-04-18:
Full reset solved the problem.

closed 2020-04-18 by raufis27

dadeboni, 2020-05-19:
I am having this issue as well after upgrading. What are you doing for the full reset?

raufis27, 2020-05-19:
> I am having this issue as well after upgrading. What are you doing for the full reset?

Install tasmota-minimal.bin (using web ui upgrade option) then flash with latest release.

dadeboni, 2020-05-19:
I have done that and i get the same behavior, I am able to control the brightness and toggle the light from the tasmota UI, but in HA the state doesnt reflect.

raufis27, 2020-05-19:
Sorry I am OH and NodeRed user. Check rules and make sure that MQTT settings are correct.

dadeboni, 2020-05-19:
Thats the strange part, rules are the exact same as posted in this wiki, mqtt settings are fine. The switch itself doesnt change the brightness either. the LEDs go up and down when pressed. Does the 8.X firmware have a different brightness command other than do dimmer like in the rule1

raufis27, 2020-05-19:
I am running v 8.2 with PWM module.

dadeboni, 2020-05-19:
Sorry for all the questions, I'm running 8.2.0.1(digiDIMv10) with the PWM module as well, Did you change anything in the rules to handle the dimming?

raufis27, 2020-05-19:
Nope, only MQTT publish. Nothing special.

dadeboni, 2020-05-19:
Figured it out, my state and command topics got flipped somewhow.. `    state_topic: "sunroomdimmer/stat/POWER"
    command_topic: "sunroomdimmer/cmnd/POWER"` instead of `    state_topic: "stat/sunroomdimmer/POWER"
    command_topic: "cmnd/sunroomdimmer/POWER"`