digiblur/Tasmota #25: bump Tasmota to 9.2

basesnow, 2020-12-17:
First of all, thank you for creating that version of Tasmota, I use it extensively with my MJ switches. turns out 9.1 as an issue that causes Tasmota to stop working and require a reboot after ~25 days. 9.2 was just released and fixes this error.

Is it possible to update your version to 9.2? Thanks!

https://github.com/arendst/Tasmota/pull/9995

digiblur, 2021-01-14:
I'll update it as soon as I see a chance to do so.

closed 2021-01-14 by digiblur

basesnow, 2021-01-14:
Awesome! Much appreciated

digiblur, 2021-01-18:
9.2.0.3 is done.

33-viper-33, 2021-01-26:
Thanks for all you do Digiblur!  upgrading to 8.5.1 minimal then to 9.2 seems to have broken all of my dimmers to include Tessan and MJ-SD01 dimmers.  On the MJ-SD01, I re-entered all your rules and enabled them and can get the dimmer to turn on and off by button and web interface.  The light will turn on with the dim up and dim down button, but will not change brightness.  The web console slider and Alexa can change brightness.  There appears to be something wrong with the dim up and down button functionality.  Any suggestions?