digiblur/Tasmota #26: Dimmers broken after upgrade from pre 8.5 to 9.2

33-viper-33, 2021-01-26:
Thanks for all you do Digiblur! upgrading to 8.5.1 minimal then to 9.2 seems to have broken all of my dimmers to include Tessan and MJ-SD01 dimmers. On the MJ-SD01, I re-entered all your rules and enabled them and can get the dimmer to turn on and off by button and web interface. The light will turn on with the dim up and dim down button, but will not change brightness. The web console slider and Alexa can change brightness. There appears to be something wrong with the dim up and down button functionality. Any suggestions?

One addition, I am currently on 9.2.0.3 of your firmware...

digiblur, 2021-01-26:
Is setoption68 set to zero?

33-viper-33, 2021-01-26:
No it wasn't.  That fixes it.  Sorry for a dumb post.  Not sure why that would have changed in with an update.  I even tried restoring from backups and it didn't correct the issue...

digiblur, 2021-01-26:
Seen it before and no idea why it turns on. I have it in a lot of my setup docs. I thought I had it in this one. Let me check.

digiblur, 2021-01-26:
And all good... No worries

closed 2021-01-26 by digiblur

33-viper-33, 2021-01-26:
Thanks DigiBlur for the help and the quick response!  All seems to be working now.

digiblur, 2021-01-26:
> Thanks DigiBlur for the help and the quick response! All seems to be working now.

No problem. I added it to the setup doc too for future people that might run into it.