digiblur/Tasmota #27: Wiki: Smart Bulb

Sineos, 2022-08-22:
I stumbled over https://github.com/digiblur/Tasmota/wiki/Smart-Bulbs#rgbw-lights-4-channel-lights because this is exactly what I'm searching. Unfortunately the used `white_*` are all deprecated in current HA versions.
Do you happen to know how to do this in todays HA? Many thanks in advance

digiblur, 2022-08-22:
Most of that is probably deprecated now and some content will move to the new website at digiblur.com as it is wiki style in a better format.  I would just use the Tasmota HA integration as it is all automatic now.

closed 2022-08-22 by digiblur

Sineos, 2022-08-23:
Thanks for your comment.
Maybe my fault but the Tasmota HA integration is not doing what I would like to:
* Default integration creating separate switches for the color channel and the white channel. 
* I would like to have both controls under one entry, e.g. color-picker for the color and slider to set. My current implementation in Node Red looks like this

[image]

I have not managed to get this done in HA. I'd appreciate any pointer.

digiblur, 2022-08-23:
You can do the split thing with Tasmota and split each channel into separate lights.  I think that is Setoption37 or something near that, I don't use it too often to keep it fresh in my memory.

Sineos, 2022-08-23:
This is exactly the point. I did this in Tasmota because I want to be able to control the white and rgb channel separately. The tasmota side works nicely but I cannot get the HA side to display the device similar to the screenshot above.

The https://github.com/digiblur/Tasmota/wiki/Smart-Bulbs#rgbw-lights-4-channel-lights example does exactly this but seems to use deprecated functions. Now I'm looking to achieve exactly this in "modern" HA but I'm failing sofar.

digiblur, 2022-08-23:
You would probably have to do some advanced mqtt template stuff to accomplish this. Not sure I understand the use case here as you can control each channel via Tasmota or mqtt without it split using color channel commands.

Sineos, 2022-08-24:
I'll try to figure it out. Likely I'm more the Node Red guy than HA. YAML and Jinja feels like, I'd shoot my own foot.
Anyway, thanks for your input. Appreciated.